etherkit/Si5351Arduino #10: xtal_freq .vs. correction

rcallen000, 2015-03-30:
I have measured my crystal freq and entered it at the init( invocation.  I expected that any error there would be the same at all frequencies.  So I went back to 25.000000 MHz and set the correction to zero.  The 10.000000 output was 10.000174 so I set the correction (unfortunately in Hz not cHz) and the freq was corrected at 10 MHz.  When I switch to 20 MHz the error was 2x of 10.  At 40MHz it was 4x and at 5 MHz it was 1.2x.  I thought I read somewher that when I put in the correction it would correct all frequencies.

So, I thought since the correction was only in Hz, not cHz, I would add my own correction.  That correction was added to the desired frequency and again worked fine at one freq but did not track linearly as the freq was changed.

I hope someone can set me straight here.  I used a similar technique with ad98xx parts and it worked fine.

Richard W5SXD

rcallen000, 2015-03-30:
1/2x at 5 MHz not 1.2x ...

NT7S, 2015-03-30:
The correction factor is applied to any frequency entered, but any remaining error will be scaled based on the relationship between the output frequency and the reference oscillator frequency. That's just the nature of how a PLL works.

Yes, at VHF the error is approximately 15x what it is at 10 MHz. So 1 Hz uncertainty would translate into a potential of +/- 15 Hz at 150 MHz.

A higher resolution could be put into the correction factor to give a more accurate calibration, but you'll never totally get away from the effect that the error scales in proportion to the output frequency. However, I would think that 0.01 Hz resolution in the correction factor would be sufficient for almost all needs.

I'd love to hear if there are others looking for a higher resolution calibration.

NT7S, 2015-04-05:
OK, correction factor is now specified in parts-per-billion, so that should give extra precision for those of you with very good frequency counters.

closed 2015-04-05 by NT7S