etherkit/Si5351Arduino #21: 5351 init - additional capacitance

akadamson, 2016-03-16:
I was working with the 5351 and a TCXO and came across a potential issue with your library.

There are some reserved bits in the additional capacitance register that need to be preserved according to the manual programming document from Si.

0b010010 = 0x12 needs to be in the low order bits *before* you add or change the additional capacitance.

I believe that the section of code at the beginning of the init routine should be modified as follows.

```
    // Set crystal load capacitance
    uint8_t tload = 0x12; // 0b010010 reserved value bits
    tload |= xtal_load_c;
    si5351_write(SI5351_CRYSTAL_LOAD, tload);

```
Also, with a TCXO, if you are just using the XA osc input, you shouldn't need the additional capacitance and so, while not documented, in review of the SI clock builder app, it appears that a value of 

`#define SI5351_CRYSTAL_LOAD_0PF       (0<<6)
`
Would be of use here.  That line would be added to the 5351.h file.  

Note, I've not tested either yet, but will be later, but fairly sure they are needed and correct.  Please see the programming manual for references as needed.

akadamson, 2016-03-16:
FYI, I confirmed no negative effects from these changes.. please verify and include as you'd like.  Thanks again for a great library.

NT7S, 2016-03-16:
Thank you much. I will try to confirm here in the next day or two.

NT7S, 2016-03-18:
@akadamson Thanks again for the find on that register. I've been running tests on that code change today. I have tested writing a 0 into the crystal load register as you have described, and it does definitely have an effect on a Si5351 with a crystal ref osc, although the effect I've seen on the TCXO is minimal. Still would be a nice thing to have in the library, so I'm testing that some more as well. In AN619, they specify a value of 0 as reserved, which is kind of odd, but not out of character with the documentation of the Si5351.

Can you give any more details of how you figured this out in the Clock Builder app? Is there an option for 0 pF in it? I just want to be sure I'm specifying the correct information before inserting it into the library.

NT7S, 2016-03-20:
Fixed in v1.1.2

closed 2016-03-20 by NT7S