etherkit/Si5351Arduino #23: pll_calc()

akadamson, 2016-04-13:
I keep looking at this and I can't figure out what it's used for.  Seems to just be code taking up space?

Middle of the above named function there is this code 

```
    // Find best approximation for b/c = fVCO mod fIN
    denom = 1000ULL * 1000ULL;
    lltmp = freq % ref_freq;
    lltmp *= denom;
    do_div(lltmp, ref_freq);

    rfrac = lltmp;
```

In the ARM compiler this tells me that rfrac is initialized but never used.  I actually comment out that last line with no ill effects... but it dawned on me that the entire block above it isn't used for anything, but is doing some nasty 64bit math and taking up those cycles?

Am I missing something?

NT7S, 2016-04-14:
Thanks, I'll double-check that. It's possible that's some cruft left over from when I initially based the library from the Linux kernel code that I found.

NT7S, 2016-04-16:
@akadamson I do believe that your analysis is correct. That code was used at one point, but when the PLL algorithm was changed, I neglected to remove it. I want to be extra certain of the impact before I remove it from production code, but I'm going to take it out of my dev branch to make sure that it has no negative impact.

NT7S, 2016-05-01:
Slated for removal in 2.0.0

PR #26 (2016-12-01): v2.0.0 Update
- Complete rewrite of tuning algorithm
- Add support for setting CLK6 and CLK7
- Add support for VCXO (on Si5351B)
- Change interface of init() and set_freq()
- Add set_freq_manual() method
- Add reset() method
- Added many new example sketechs

Fixes #23 and #24
closed 2016-12-01 by NT7S