etherkit/Si5351Arduino #24: CLK1 & CLK2 Frequency Setting Interaction Errors when Sharing PLLB

NT7S, 2016-08-21:
The frequency setting of either CLK1 or CLK2 outputs significantly impacts the frequency of CLK2 or CLK1 respectively by 100 KHz or more when operating at HF frequencies. The CLK that is set last is accurately set but the previously set CLK sharing the same PLL is frequency shifted.

This problem only affects CLK outputs sharing the same PLL. Since there are two PLL's the issue is only significant for applications where 3 outputs are needed as two of the outputs must share a PLL.

(copied from @va7ta report on the AVR sister library https://github.com/NT7S/Si5351/issues/11)

NT7S, 2016-11-06:
Initial v2.0.0 commit in libupdate branch includes a new tuning algorithm which addresses this problem.

va7ta, 2016-11-07:
Hi Jason,

     I received this notification - many thanks for following up on this 
bug!

     I decided to check out your new V2.0.0  but couldn't find it. 
Arduino Library Manager update shows latest version as 1.1.2 and github 
zip files are dated last June.

     Does it usually take a while for the updates to show up on 
distribution sites?

     If you would like me to take it for a spin before it shows up in 
distribution please send me the files.

Best Regards,
Tom, VA7TA

NT7S, 2016-11-07:
Hi Tom,

You can find the branch here: https://github.com/etherkit/Si5351Arduino/tree/libupdate

You'll have to manually install the ZIP file in your library folder for now. It's still not ready to be published because I have more features to add, but it is definitely in a state that is suitable for testing with the A variant.

Please do take it for a spin and kick the wheels. I'm sure there are probably a few more bugs to work out. Please notice that the interface has changed a bit, and may change more in the future before release, so look a the README for the updated documentation.

Thanks,
Jason

va7ta, 2016-11-07:
Hi Jason,

      Many thanks!

     Just to let you know your new si5351example.ino file failed to 
compile because:

- clock frequency CLOCKFREQ middle parameter needs to be added to line 30

#define CLOCKFREQ 25000000

- setfreq lines 33 and 37  have too many parameters - middle parameter 
needs to be removed to match new lib, I think.

     Compiles OK now. Hope this is helpful.

Tom

NT7S, 2016-11-07:
Sorry about that, I know that the example sketches are not in a good state yet, but I forgot to mention it. I'll get them tidied up very soon.

va7ta, 2016-11-08:
Hi Jason,

     Nothing to be sorry for - I understood that this is a work in 
progress - just wanted to let you know what I spotted.

     Gosh you write nice magic!!!

     I made the necessary changes to my WiFi server code to match your 
new lib and I am happy to report that the 5351 output frequencies are 
now all precise! I calibrated the 5351 set to 100 MHz against my counter 
which I have connected to an external frequency standard and got it to 
within a few Hz. Then I found the other two outputs which were set to HF 
frequencies were both within 1 Hz.

     Many thanks for your great work! Hopefully I will get another 
notification from github when you have it all finalized.

Thanks Again!
Tom, VA7TA

NT7S, 2016-11-12:
Glad that it looks like it solves that nasty problem. I've got a little bit more code to add to this version, then I'll make an announcement of a beta version. Hopefully in no more than one week.

closed 2016-12-01 by NT7S