etherkit/Si5351Arduino #35: factory programed Si5351

lz1zp, 2017-02-11:
Hello Jason,
I am using your Si5351 library for several of my HAM projects and it's a "great job" - THANKS a lot! 
Was not informed about different versions of Si5351A-Bxxxx-GT and bought factory programed chips, which does not want to work with previous wrote applications !? Examples of library generates the following message on monitor:
SYS_INIT: 0  LOL_A: 1  LOL_B: 1  LOS: 0  REVID: 0
and CLK0...CLK2 are generating frequencies programed into NVM....
My question is - how I can reset and overwrite registers with my code ? In other word - how i can use the same code with blank and factory programed Si5351 using your labrary ?
Best regards,
Georgi

NT7S, 2017-02-17:
Hi Georgi,

Without having that custom ICs that you have it would be difficult to figure out exactly how to solve your problem. However, you can access the raw commands for reading and writing registers to the Si5351, so what you may want to do is first write a sketch to read the contents of each register using the si5351_read() method and then printing them to a serial terminal. Then you should be able to compare to a plain Si5351 or maybe even the reported default values in application note AN619. After that, you can try using si5351_write() to see if you can get it to a state where you can use it as desired. If you are able to find out what needs to be done, let me know and I may be able to incorporate some code into the library.

lz1zp, 2017-02-18:
Hello Jason,
Thanks for your reply and for advise what to do.
First of all, I have to inform you, I have tried to change the Si5351.cpp
file depending to data-sheet page18 and page 19 - NVM storing into RAM
during power-up and following custom configuration via I2C, but it does not
helps me, or just I didn't do the right changes. I have tried to modify
si5351::reset(void)....
Now I involved in this "problem" a friend of mine LZ1NUL, who is a
programing engineer . Will inform you at the end - for "good" or "bad" !

73's
Georgi

NT7S, 2017-03-17:
Suspect the problem may be because the IC has an I2C address other than 0x60. The ability to init the library with alternate I2C addresses has been added to v2.0.2, so I'm going to close this for now unless it can be shown that there is some other issue.

closed 2017-03-17 by NT7S

lz1zp, 2017-03-17:
Hello Jason,
It's a shame I can't test it right now -  left this IC at my job.....Will
try to do that tomorrow or on Monday.

Thank you in advance, 73's
Georgi

lz1zp, 2017-03-20:
Hello Jason,
It's working !! Now I can program factory programmed Si5351, same way as
they are blank. Difference is exactly into the I2C address - 0x60 blank
against 0x62 factory OTP. Now with "Si5351 si5351(0x62);" initialization I
can use the same code....
Tested set_freq() method up to to 225 MHz - working great.

Congratulations about everything you have done on Si5351 library, examples
& etc.Thank you very much for your support.

Best regards,
Georgi LZ1ZP

2017-03-17 22:46 GMT+02:00 Georgi Nestorov <nestorov.georgi@gmail.com>:

NT7S, 2017-03-24:
Excellent, glad I could be of assistance