etherkit/Si5351Arduino #36: Two si5351s on the same I2C bus

prt459, 2017-02-21:
In the interests of separating three clocks in a receiver project, I'm considering a configuration with two si5351s on the same Arduino controller and I2C bus.  Currently Wire.BeginTransmission() uses SI5351_BUS_BASE_ADDR (x60). If this #define constant was changed to a private attribute that defaulted to SI5351_BUS_BASE_ADDR, and a public setter added, I could explicitly set the target device's address before making library calls.   And 99% of the library's users with only one si5351 would never know!  Just a suggestion.

NT7S, 2017-02-25:
This sounds quite reasonable and I'll see if I can get it in the next release. Thank you for the suggestion.

NT7S, 2017-02-25:
@prt459 OK, code pushed to dev tree: https://github.com/etherkit/Si5351Arduino/tree/v2.0.2

If you could please install this version of the library manually and give it a try I would appreciate it. I put a bit of info in the README to show how it works, but just add the desired I2C address to the class constructor at init.

closed 2017-03-17 by NT7S