etherkit/Si5351Arduino #37: Hang on init

michelinok, 2017-02-22:
Hi!
i'm trying a simple sketch...but it hang (no error in compiling sequence) when i init the si5351

si5351.init(SI5351_CRYSTAL_LOAD_8PF, 0, 0);

the arduino just sit there.
Any idea?

DallasEpperson, 2017-02-23:
Are you sure you have the si5351 connected to the arduino using the correct pins? If not, the i2c connection won't establish.

michelinok, 2017-02-23:
Yes, i'm sure it's connected in the right way because if i use the adafruit example it works.
I've used this tutorial:

https://learn.adafruit.com/adafruit-si5351-clock-generator-breakout/wiring-and-test

And it works.

Any help is really appreciated.

michelinok, 2017-02-23:
Ok...found that the problem is instead on jtencode.jt65_encode(message, tx_buffer);
I'll open a ticket on jtencode library to explain

closed 2017-02-23 by michelinok