etherkit/Si5351Arduino #42: what is the default drive strength at startup?

amunters, 2017-03-29:
Not an issue, but just a few questions:

It is possible to set the drive strength, for example to 4mA:
si5351.drive_strength(SI5351_CLK0, SI5351_DRIVE_4MA);

Questions:
1. What is the default drive strength at startup condition?
(what is de drive strength when the command si5351.drive_strength() is never issued at all)?
2. What parameter values for drive strength are accepted?

Allard PE1NWL

NT7S, 2017-03-29:
As stated in application note AN619, the default value of that register is 00, which corresponds to 2 mA. This library doesn't change that upon initialization.

The drive level enumeration values are listed in this section of the documentation: https://github.com/etherkit/Si5351Arduino#tokens

closed 2017-03-29 by NT7S

amunters, 2017-03-29:
Thanks Jason! All understood now.
Perhaps it would be a good idea to add this to the section "Startup Conditions" in the README.md?
Thanks for this great library!
73, Allard PE1NWL

NT7S, 2017-03-29:
Yes, I've already put it in the README in the next release, thank you.