etherkit/Si5351Arduino #44: Interrupted clock during tuning changes

NT7S, 2017-04-10:
When using the library in a receiver, there are reports that the CLK output has a discontinuity when the frequency is changed, causing an audible popping in the audio.

It is known that this is the result of the PLL reset in the tuning code, which is sometimes needed to ensure that the output is on the correct frequency. What is needed is a way to minimize these PLL resets to avoid this problem as much as possible.

la3pna, 2017-04-10:
How about doing it like some of the Si570 libraries does it, where if the
change deltaF is less than a given value, lets say 10KHz you do not
reset?this way, the tuning should appear continous for small changes where
its clearly visible, yet for larger changes, and bandswitching, the
discontinuty would be once.

NT7S, 2017-04-10:
I need to do some testing into exactly when it is required for the PLL to be reset in the course of changing the multisynth parameters. I believe at a minimum it should be feasible to suppress PLL reset across an amateur radio band, and perhaps under even more circumstances.

sp9mrn, 2017-04-10:
but why ver 202 works without clicks?

MAc
sp9mrn

NT7S, 2017-04-10:
@sp9mrn The PLL reset wasn't in the set_freq() method previous to that (well it was in an earlier version but had been removed for a while):

https://github.com/etherkit/Si5351Arduino/commit/8c2c45b228417b7320595b22a1f1a4402cf642f8#commitcomment-21702858

sp9mrn, 2017-04-10:
" I believe at a minimum it should be feasible to suppress PLL reset across an amateur radio band" - yes, but we have to describe condition "across band" without wasteing time for calaculating it (the frequencies) from IF&VFO.
What does it mean for controller "you are still inside band" without checking "changeband' state?
We can of course chceck "band NR" or "changeband" variable, but it is good for sketch, not for library.
Thanks for Your work ;-)
MAc

pavelmc, 2017-04-11:
Please visit and test this lib https://github.com/pavelmc/Si5351mcu it does not have click noise.

The trick is to just make a reset of the PLLs just when it's needed; and that is at the initialization on a sweetsop frequency (in the setup() procedure) then you recalculate the PLLs and msynths and just update them, it will work 100%

Also my correction procedure do make click noise as I have to reset the IC after any XTAL freq change

I have published it yesterday and had it test about ~1 month now in this project: https://github.com/pavelmc/arduino-arcs with click free operation in my bench and tested and confirmed the click-less operation by at least 2 more users. (The lib was made up in the last weekend, the code is embedded on the project files since a few time now) 

The test I made has revealed (see lib example) that I can move the freq from 25 to 150 Mhz without resetting the PLL/msynths and it hasn't any secondary effect against the same run resetting on every freq change (beside click-less operation)

I has made it run trough 1 to 10 Mhz also and it perform the same with/without resetting.

73 de Pavel CO7WT.

PS: There is at least one report that the lib do has click noise, but it may be from hardware setup conditions, I'm digging about it and waiting for more reports to confirm, the lib has just one day of life.

NT7S, 2017-04-19:
I just pushed a branch where all I did was comment out the PLL reset. I'd like to see if there are instances where tuning fails under these circumstances, so if those of you having tuning problems can please try this out and let me know how it works, it would be appreciated. To try it out, manually install the library in your Arduino installation.

https://github.com/etherkit/si5351arduino/tree/v2.0.5

dgbalharrie, 2017-04-21:
Works with no clicks for me - thanks for the effort
73 De M0DGB

sp9mrn, 2017-04-21:
Sound of silence. Wonderful.
 I checked only in range of 300kHz. I suppose I can call PLLreset with "bandchange" command if necessary.
Thank You

NT7S, 2017-06-17:
Fixed by #48

closed 2017-06-17 by NT7S