etherkit/Si5351Arduino #47: Si5351A- 3 CLK outputs

Silhouette-Zed, 2017-05-04:
Hey, 

Firstly, I am an amateur of GitHub and I really thank for the library. I am using the library for a Si5351 GUI project, and my Si5351 has 3 CLK outputs, but when I try to arrange the library for CLK0 and 1 to share PLLA and CLK2 to use PLLB, I am failing can you help, please.

NT7S, 2017-05-04:
Can you please describe the nature of the failure in more detail and post the code being used?

Silhouette-Zed, 2017-05-04:
Thanks for reply, 

I am simply using the serial port for communication. Firstly, I changed the library a little bit for making CLK0 and CLK1 to use PLLA and CLK2 to use PLLB. I am not really sure that I achieved. I believe the problem is with the library being not suitable for 3 Clocks.

I first did this,

```
`pll_assignment[0] = SI5351_PLLA;
pll_assignment[1] = SI5351_PLLA;
pll_assignment[2] = SI5351_PLLB;`
```

then this,

```
`set_ms_source(SI5351_CLK0, SI5351_PLLA);
set_ms_source(SI5351_CLK1, SI5351_PLLA);
set_ms_source(SI5351_CLK2, SI5351_PLLB);
...
for(i = 0; i < 3; i++)
{
	clk_freq[i] = 0;
	output_enable((enum si5351_clock)i, 0);
}
`
```
Also, the if condition in line 199

`if((uint8_t)clk <= (uint8_t)SI5351_CLK1)
`

There is few more editing like this. First problem is that when I set clock 1 or 2 they are being set to same frequency. 

Second problem is that I tried to use the invert property of the library, anyhow, when I try to invert the clocks, the signal just get losts. Clock number comes through the serial port. I put bunch of stars* to the problematic parts, but I dont think there is a problem with them, 

```
`#include "si5351.h"
#include "Wire.h"

Si5351 si5351;
int led = 8;

char mode, inData;
String clkStr, freqStr, Str, strengthStr, corrStr, pll_freqStr, ratioStr;
String startFreqStr, stopFreqStr, stepSizeStr, stepTimeStr, analogPinStr, sweepType;
int clk, strength, ratio;
int analogPin = 0;
unsigned long freq, startFreq, stopFreq, stepSize, stepTime;
long corr;

uint64_t target_freq = 1000000000ULL; // 10 MHz, in hundredths of hertz
unsigned long long freqP, freqq;
unsigned long long pll_freq, pll_freqq;

void setup() {
  Serial.begin(9600);
  pinMode(led, OUTPUT);
  digitalWrite(led, LOW);
  si5351.init(SI5351_CRYSTAL_LOAD_8PF, 0, 0);
//  si5351.set_freq(1000000 * SI5351_FREQ_MULT, SI5351_CLK0);
//  si5351.set_freq(200000000ULL, SI5351_CLK1);
  si5351.update_status();
  delay(500);
  glimpse(1);
}

void loop() {
  while(Serial.available() == 0){}
  mode = Serial.read();
  
[... 285 lines cut ...]

elseif index_selected == 2
    fprintf(s,'b%s',get(handles.freq1, 'String'));
    pause(0.2);
    fprintf(s,'a%s',num2str(val));
    pause(0.2);

elseif index_selected == 3
    fprintf(s,'c%s',get(handles.freq1, 'String'));
    pause(0.2);
    fprintf(s,'a%s',num2str(val));
    pause(0.2);

elseif index_selected == 4
    fprintf(s,'d%s',get(handles.freq1, 'String'));
    pause(0.2);
    fprintf(s,'a%s',num2str(val));
    pause(0.2);

else
    warndlg('Not a proper option.','Error')
    fprintf('Error in set_freq0. \n');
end`

Thanks in advance, and sorry for messy coding.

Silhouette-Zed, 2017-05-10:
I solved the problem, thanks. If you want I can add my GUI as a sample project.

closed 2017-05-10 by NT7S