etherkit/Si5351Arduino #65: pll not recalculated ( 150 MHz boundary ?? )

gartnl, 2018-07-11:
Right after initializing ( so with default values for pll-source, plla )

Consider:
```
si5351.set_freq(8000000000ULL, SI5351_CLK0);
si5351.set_freq(15600000000ULL, SI5351_CLK0);
si5351.set_freq(8000000000ULL, SI5351_CLK0);
```
The second 80 MHz is not put out. Some debugging ( looking at the pll value ) showed that the pll is not recalculated.
Now this:
```
si5351.set_freq(8000000000ULL, SI5351_CLK0);
si5351.set_freq(15600000000ULL, SI5351_CLK0);
si5351.set_freq(14300000000ULL, SI5351_CLK0);
si5351.set_freq(8000000000ULL, SI5351_CLK0);
```
This works. The 143 MHz triggers recalculating the pll.

Workaround:
```
si5351.set_freq(8000000000ULL, SI5351_CLK0);
si5351.set_freq(15600000000ULL, SI5351_CLK0);
si5351.set_freq(8000000000ULL, SI5351_CLK0);
si5351.pll_reset(SI5351_PLLA);
```
Pll is recalculated and the 80 MHz is put out.

I can live with it, but took me some time to figure out.
Is this the si5351 or a glitch in the pll calcualtion?

la3pna, 2018-07-11:
What verson of the library? I'm not able to reproduce.
AVR or ARM? there may be a compiler issue here.  I have only tried on ARM.

FWIW, I have not had any issues with signals above 150MHz, the chip works
quite ok to 225MHz, where a internal register overflows.

Jason is away for a week or so, he will probably reply when back.

Thomas.

gartnl, 2018-07-14:
@ Jason
Seems indeed to be two different things.

The first part about needing a pll-reset when doing a switch from F>150 to F<100 stands. 

The second part is rather strange, and seems not to involve the library, so you can forget about it. Just FYI: 

```
si5351.set_freq(15500000000ULL, SI5351_CLK0);
si5351.set_freq(9900000000ULL, SI5351_CLK2);
```

does not work for CLK2, but

```
si5351.set_freq(18000000000ULL, SI5351_CLK0);
si5351.set_freq(9900000000ULL, SI5351_CLK2);
```

is just fine.

Turns out that whenever the multisynth divider is between 6 and 6.8 my si5351 does not output a signal, or a strangely mutilated signal. An integer value of 6 is ok however. eg If I set a pll of 600 , 100 MHz is ok, but 98 is not ? Even when according to AN619 any divider between 6 and 1800 should be ok.
This gives a limitation that I cannot find in the datasheet: if Fclkx > 150 , the other clocks on the same pll can have a maximum frequency of (Fclkx*4)/6.8 .  Strange ..

closed 2026-10-10 by NT7S