etherkit/Si5351Arduino #66: library integration on ESP32

micfuc, 2018-11-20:
Hi , I am integrating Si5351Arduino on an ESP32 micro using arduino IDE.
I get erratic behaveures when using the set_freq and set_freq_manual methods; in particular:
1) the status is always as following
SYS_INIT: 1  LOL_A: 1  LOL_B: 1  LOS: 1  REVID: 0
2) when setting more then one CLK with set_freq   no out is generated; with a single clk it works
3) when setting frequencies > 10 Mhz the actual value obtained is completely wrong
4) by using the set_freq_manual  it works pretty good on  a single clock
5) by using the set_freq_manual  with 2 or 3 clocks does not work...

can you please help to fix these issues ?

Thanks in advance for any help and thanks for the wonderfull work done ...

Mike

NT7S, 2018-12-18:
I recently acquired an ESP32, so at some point I should be able to look at this, but right now I'm in the middle of a crunch to get some other projects shipped, so this will have to wait a bit. Thanks for the report.

paulh002, 2019-07-03:
I have exactly the same issue. I now use a interface developed by JF3HZB maybe there is some timing issue

WA2FZW, 2019-08-15:
Me three! I just reported this to Jason directly yesterday. Unlike Mike, however, when I look at the status bits, they all look good! The program I'm trying to make work is the "si5351_calibration" program from the examples that come with Jason's library. The only mods I've made to it (other than a million different approaches to trying yo make it work) was to actually read the frequency and PLL frequencies from the Si5351 to display them instead of just displaying what the program thinks they are.

I can also confirm that the JF3HZB  software works as I have a modified version of his VFO working, and if I could figure out how to add the calibration function to it, I would!

John

WA2FZW, 2019-08-15:
I tried the "si5351.set_freq_manual" function this morning - didn't help! Also tried different frequencies (5MHz, 10MHz and 15MHz) - Still just get a short burst of output then the Si5351 stops outputting. I'm only using CLK0 and PLLA, again in a slightly modified version of Jason's calibration program.

The output I get is:

Si5351 Calibration Program
Waiting for Si5351 to come online!

Si5351 is online!

I2C Bus Speed:      100,000.00 Hz
Actual correction:  0.00 Hz                              // Have this commented out
Target Frequency:   15,000,000.00 Hz             // What the program thinks
Actual clock freq:  15,000,000.00 Hz               // What the chip thinks
PLL-A Frequency:    800,000,000.00 Hz          // What the chip thinks
PLL-B Frequency:    800,000,000.00 Hz          // What the chip thinks

SYS_INIT: 0  LOL_A: 0  LOL_B: 0  LOS: 0  REVID: 0

Adjust until your frequency counter reads as close to 10 MHz as possible.
Press 'q' when complete.
   Up:   r   t  y  u  i   o  p
 Down:   f   g  h  j  k   l  ;
   Hz: 0.01 0.1 1 10 100 1K 10k

Everything except "Target Frequency" is actually being read out of the chip.

And BTW, I'm using an Adafruit board; I see no significant electrical difference between the Adafruit board and the Etherkit board.

My current
[Si5351_Cal_With_Alternate_I2C.txt](https://github.com/etherkit/Si5351Arduino/files/3505972/Si5351_Cal_With_Alternate_I2C.txt)

The current version of my code is attached (change the extension from "txt" to "ino")

John

WanderLG, 2019-09-10:
Most likely the problem is in the ESP32 Wire library, because using a logical parser, we can see that ESP32 loses I2C SCL sync.
I am sending 2 files with si5351_calibration.ino i2c logical data
Arduino vs ESP32

If you want to verify both files, download the Saleae Software.
https://www.saleae.com/pt/downloads/ 

Muito provavelmente o problema é na biblioteca Wire do ESP32, pois usando um analisador logico, podemos perceber que o ESP32 perde o sincronismo SCL do I2C.
Estou enviando 2 arquivos com os dados lógicos i2c do si5351_calibration.ino
Arduino vs ESP32

Se quiserem verificar os dois arquivos, baixem o Software Saleae.
https://www.saleae.com/pt/downloads/ 

[Analyzer Logic.zip](https://github.com/etherkit/Si5351Arduino/files/3593190/Analyzer.Logic.zip)

WA2FZW, 2019-09-10:
Thanks! I didn’t download the program to view the logic analyzer files, but I’m not convinced the problem is in the “Wire” library. Aside from the problem with the Si5351 on the ESP32, I am using the “Wire” library along with a library for the PCF8574 <https://github.com/RobTillaart/Arduino/tree/master/libraries/PCF8574>  chip that works fine on the ESP32. I haven’t done an analysis to see if the two libraries are using the “Wire” library differently or not.
 
I published an Si5351 calibration program <https://github.com/WA2FZW/A-Program-to-Calibrate-the-Si5351-by-WA2FZW>  on GitHub last week which does not use the “Wire” library. It uses a slightly modified version of the Si5351 software from the digital VFO designed by TJ  <https://www.youtube.com/watch?v=HjzTwqTvjqQ&t=5s> Uebo (JF3HZB) and works equally well on either the ESP32 or an Arduino.
 
John
WA2FZW

 

https://www.saleae.com/pt/downloads/
https://www.saleae.com/pt/downloads/

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

WanderLG, 2019-09-10:
But something is happening and the package is getting lost.
look at these two comparisons.

This picture is the wire scanning for addresses. This one works perfectly.
[image]

and this is si5351_calibration.ino This does not work..

[image]

WanderLG, 2019-09-10:
The two photos are just the end of the package. In the first photo, both work, but in the second photo, Arduino Works and ESP32 does not work.

WA2FZW, 2019-09-10:
Oh, I agree something isn't working; just not sure what! You didn't say, but is the scanner program using the Wire library or not?

I definitely see some differences in the data streams in the 2nd set of pictures, but I'm not sure what they mean. There are a couple of strange pulses on the SCL line also:

[image]

John
WA2FZW

WanderLG, 2019-09-10:
All using wire.

WA2FZW, 2019-09-10:
Well, again, that would lead me to believe the problem is not in the Wire library, but in the Etherkit library. Again I am using the Wire library to communicate with a PCF8574 over the I2C bus with no problem, which would also lead me to believe the Wire library is OK.
 
John
WA2FZW

 

https://www.avast.com/antivirus

WA2FZW, 2019-09-10:
This didn't seem to get posted when I replied via email:

Well, again, that would lead me to believe the problem is not in the Wire library, but in the Etherkit library. Again I am using the Wire library to communicate with a PCF8574 over the I2C bus with no problem, which would also lead me to believe the Wire library is OK.

John
WA2FZW

ramapcsx2, 2019-09-10:
Hey,
the problem could be in the I2C implementation in the Si5351 chip itself.
I2C features like clock stretching could be used and this is known to cause issues with the ESP32 in particular.
So I recommend investigating for I2C protocol errors.
Check active and closed issues on the esp32-arduino project such as this one: https://github.com/espressif/arduino-esp32/pull/2301

Good luck!

WanderLG, 2019-09-10:
This is where the first error occurs.
Note that using Arduino, we have between 16+ACK up to 12+ACK 9 high SCL pulses.

But using ESP32 that doesn't happen and we only have 5 high pulses in SCL.

[image]

Jhon, I suspect the problem is with ESP32 i2c libraries as there are more people with this same problem using other i2c modules. Already the (JF3HZB) VFO works, because it built i2c communication without using the ESP32 specific libraries. It's not just Wire.h or Wire.cpp. There are others that are related.

WanderLG, 2019-09-10:
But yes, there may be a way to work around this in the Etherkit SI5351 library without having to touch the ESP32 i2c libraries. I just don't know how to do this. But I'll try, if I can before you, post the result.

Note: I am Brazilian and do not speak English. I'm using google translator.
Wander. (PU4WLG)

WanderLG, 2019-09-10:
There are people saying that they solved the ESP32 i2c problem by updating the package.

https://github.com/espressif/arduino-esp32/releases

Now I have no way to test this. But as soon as I get home, I'll take the test.

WA2FZW, 2019-09-10:
That would be great! Let us know if it works! And tell me how to go about doing the update! I couldn't figure it out! Thanks!

John WA2FZW

WanderLG, 2019-09-11:
I couldn't make it work. Updated to:
ESP32 Arduino 1.0.3 RC3 (IDF 3.2)
But still the error continues.
To generate frequency in only one output, CLK0 for example.
The JF3HZB / T.UEBO library works perfectly.
But it's manual so I don't know if it's possible to generate 3 independent outputs, like in this Jason Milldrum library.

VFO JF3HZB
https://onedrive.live.com/?authkey=%21APjYCcc%2DZzSw%2DCQ&id=567BE007779244E3%21101672&cid=567BE007779244E3

WA2FZW, 2019-09-11:
Wander – Email me direct at my call sign at ARRL.net. I’ll send some code that shows that you can use all 3 clocks independently with TJ’s code. I don’t want to post it on GitHub just yet as testing is not complete.
 
John
WA2FZW

 

https://onedrive.live.com/?authkey=%21APjYCcc%2DZzSw%2DCQ <https://onedrive.live.com/?authkey=%21APjYCcc%2DZzSw%2DCQ&id=567BE007779244E3%21101672&cid=567BE007779244E3> &id=567BE007779244E3%21101672&cid=567BE007779244E3

---
https://www.avast.com/antivirus

WanderLG, 2019-09-11:
Jhon Thank you. I sent the email. Did it get there?

WA2FZW, 2019-09-11:
Yes – Answering now!

 

https://www.avast.com/antivirus

paulh002, 2020-03-24:
I have replaced the wire library by direct sending the i2c commands on the GPIO so called bit banging. No it looks it works also on ESP32. You have to replace all the wire stuff. But thats relative simple. Did not test everything but for my application it is working.  Code can be found on https://1drv.ms/f/s!Ap0-o26gdN7pjPI5fhf2OSHI2jg-EQ

kd2eat, 2020-07-22:
I can confirm that there is still a Wire library problem on ESP32 with version 1.0.4 for Arduino.  I can't even get the basic Etherkit test programs to run.

I grabbed Paul's i2c and si5351 libraries above and was able to get test programs running.  I'm going to move forward with my own code, which uses the VCXO.  Paul, if I find glitches I'll post to your github.

Hopefully, Espressif will figure out their issue and fix their stock i2c implementation.

NT7S, 2020-07-22:
FWIW, I've ordered an ESP32 board so I can see what's going on for myself. If it is down to the Wire implementation for the ESP32 support package, I'll close this issue. However, I'd like this library to be as universal as possible, so it's long past time for me acquire an ESP32 board and make sure it can work.

kd2eat, 2020-07-22:
Awesome!  Thanks for your great work with this library, BTW, I've used it for several little projects and it works like a champ.  I've got a Logic analyzer and a pile of different model Arduinos and a few ESPs.  If I get motivated to take a stab at things, I'll let you know.  Also, if you need a test dummy, I'd love to give back for all your hard work.  Just let me know if I can help.

paulh002, 2020-07-22:
Some people got it working without an issue, so there is some variation. Maybe if you combine it with an ili9341 spi interface the i2c gets in problems. Anyway I also added multi si5351 support. Would be a nice extension to the library. This to lower the crosstalk between channels.

vk4hat, 2020-12-24:
The problem is something within the etherkit library, I never used to have this problem until i moved to platformio. My arduino implementation of ESP32 was old, very old and it worked flawlessly. But on anything new it fails. I have gone hunting for the issue but cant find it. 

I have other libraries for SI5351 that do work with ESP32 the one from adafruit and there is one by Pavel Milanes that works flawlessly, so my guess is that there is something not right in how the etherkit library is talking over I2C.

paulh002, 2020-12-24:
Hi Rob, in stead of moving to an other library, I changed the i2c part of
etherkit. I also changed it to support multiple si5351's. You can get the
code if you like on my github page. paulh002

By the way, I could not give comments to your blog, I think there is some
issue

73 Paul

vk4hat, 2020-12-24:
> Hi Rob, in stead of moving to an other library, I changed the i2c part of etherkit. I also changed it to support multiple si5351's. You can get the code if you like on my github page. paulh002 By [...]

Thanks Paul, I will grab your code later and see if I can manage to make the quadrature SI5351 work with it. I did fix the comments on my blog, seems Jetpack changed settings and made people login with facebook or google to comment. That might explain why i have not had comments in a long time and no one let me know LOL 

Cheers Mate

Hi-Tone, 2021-01-25:
I have a similar issue with this library (and all others who've contributed) and the ESP32. I've tried the Si5351_Cal_With_Alternate_I2C sketch by **WA2FZW** and I get an "Si5351 is online", however no frequency output. 
I've tried the Adafruit library and that works out of the box, however it not very beneficial, so I am eager to get this one working.

I've looked into the I2C issues, however I've tried with all the sketches by various people commenting on this topic, and none of them gets me any output. In any case, working with ESP32, I always have to use:
#define	SDA	   	  23
#define	SCL	   	  22

If I don't use these, together with Wire.begin(SDA, SCL); my i2c never works with any module or sensor.

My setup is PlatformIO on VScode, NodeMCU-32s and an Adafruit module (I probably would have bought the Etherkit if I was aware of it before buying the Adafruit SI5351). I am measuring my output on a Agilent 53131A.

vk4hat, 2021-01-26:
> I have a similar issue with this library (and all others who've contributed) and the ESP32. I've tried the Si5351_Cal_With_Alternate_I2C sketch by **WA2FZW** and I get an "Si5351 is online", however [...]

https://github.com/pavelmc/Si5351mcu This library does work with ESP32 out of the box. Its limited to 2 clocks only though. There is also a Russian\Ukrainian library that works as well, but dont have a link handy.

Hi-Tone, 2021-01-26:
I got it somewhat working now. I used the library by **pavelmc** and by setting the standard I2C pins for ESP32 (SCL - pin 22; SDA - pin 21). I used the simple example, however at 60 MHz I get close to 55 MHz output. At 112.5 MHz, I get 104 MHz output. Is it normal for it to deviate that much? 
With the Adafruit library, I am able to get get 112.5 MHz by setting the 112.5MHz output.

PS: I tried afterwards the sketch from WA2FZW without any luck :(

vk4hat, 2021-01-26:
> I got it somewhat working now. I used the library by **pavelmc** and by setting the standard I2C pins for ESP32 (SCL - pin 22; SDA - pin 21). I used the simple example, however at 60 MHz I get close [...]

Reset the PPL's

Hi-Tone, 2021-01-27:
I assume you mean the PLL? In the library you attached by **pavelmc**, this is already done as standard.

vk4hat, 2021-01-27:
> 
> 
> I assume you mean the PLL? In the library you attached by **pavelmc**, this is already done as standard.

If you are making a jump of more than a meg or so, it needs to be called again. I do not have any problem with it being inaccurate.

kd2eat, 2021-02-15:
Just an update on this problem.  I tested the Etherkit library again with Espressif's Arduino esp32-1.0.5-rc6 build.  Still no joy.  The behavior seems about the same as with the 1.0.4 release.  I was hoping the vendor lib's might fix it, since there have been updates in the wire code.  Apparently not.  :-(

I'm having a few minor issues with Paul's "bit-banging" library.  It doesn't seem to play well if I'm using I2C via the Wire library on the same pins.  I either have to figure out why, or redesign my riser board to bit-bang on a different pair of GPIO ports for using the Si5351.  Paul's also doesn't seem to disable transmit when I call:
     si5351.output_enable(OUTPUT_CLOCK, SI5351DISABLE);
I may dig into that and see if I can sort it out.  That's actually more problematic than the Wire() conflict at the moment.

Anyway, I thought I'd share an update.
Mike

paulh002, 2021-02-15:
Sharing the same ports with Wire can indeed be a problem, in my project I only have si5351's attached. The output_enable function I did not use. But i will check. It is a pity that the wire lib does not work, I tried to look into it did some analysis with my scope that can decode i2c but no luck. Cost too much time to debug this issue.

kd2eat, 2021-02-15:
Thanks Paul, I read through the code a bit last night and everything looked about right, though I didn't do any serious debugging.  I was hoping it was something as simple as you hadn't implemented the routine and just stubbed it in.  

I'll try to hook up my logic analyzer in the next few days to get a capture of the behavior.  If you want to take this offline, since it's not really related to the etherkit library, feel free to contact me at kd2eat at gmail.com.  Thanks for writing back :-)

davekw7, 2021-06-18:
Here's what works for me:

In the SI5351::si5351_read method

Change
`  //Wire.requestFrom(i2c_bus_addr, (uint8_t)1, (uint8_t)false);`

to
`  Wire.requestFrom(i2c_bus_addr, (uint8_t)1);//, (uint8_t)false);`

Arduino 1.8.13

ESP32 package 1.0.6

Windows 10

Regards,

Dave

paulh002, 2021-06-18:
That would be great, will test it

paulh002, 2021-11-20:
> Here's what works for me:
> 
> In the SI5351::si5351_read method
> 
> Change ` Wire.requestFrom(i2c_bus_addr, (uint8_t)1, (uint8_t)false);`
> 
> to ` Wire.requestFrom(i2c_bus_addr, (uint8_t)1);//, (uint8_t)false);`
> 
> Arduino 1.8.13
> 
> ESP32 package 1.0.6
> 
> Windows 10
> 
> Regards,
> 
> Dave

I have tested this solution and it works good. 
Thanks

PE2BS, 2022-11-09:
Thanks, works well. It was the solution for the problem with the ESP32 and the Si5351 board. 
73
Ruud / PE2BS

vk4hat, 2022-11-09:
Awesome. Happy that you got it working now. If you run into any other 
problems, just sing out, I will try and help as best i can over email

---
********************************************************************************
   Rob Powell, VK4HAT.
   Blackbutt, QLD, Au.
   http://robs-blog.net
********************************************************************************

Links:
------
[1] 
https://github.com/etherkit/Si5351Arduino/issues/66#issuecomment-1308774680
[2] 
https://github.com/