etherkit/Si5351Arduino #70: Si5351B 127mhz<not working<150mhz

hakkican, 2019-06-08:
i am using **si5351b** and 25mhz crystal. I can't see the output between 127 MHz and 150 MHz.
`si5351.set_freq(14600000000ULL, SI5351_CLK0);`
and
`si5351.set_ms_source(SI5351_CLK0, SI5351_PLLB); // or SI5351_PLLA
si5351.set_freq_manual(14600000000ULL, 87600000000ULL, SI5351_CLK0);`
I tried both methods. 
is this an bug? What am I doing wrong? (v2.1.4 )

la3pna, 2019-06-09:
Are you sure you have all the power rails correct? What does the Si5351 return? Try the demo sketch with a change of frequency to be sure. 

Got a B version here working correctly. 

Br. 
Thomas. 

Sendt fra min iPhone

hakkican, 2019-06-09:
I made the connections with the Si5351BCBreakout.pdf file. I'm using demo sketch. I've tried changing frequencies. I see the signal at 127mhz. But no signal between 128-149mhz. And I received the signal again on 150mhz.

NT7S, 2019-06-10:
https://nt7s.com/2019/06/regarding-recent-availability/

zanco, 2020-08-15:
Hi,
Can I add to this issue that for me (SI-5351A-B-TG) the same problem occurs ?
When running an oscillator.ini which does not use any libray but writes to the registers I can have it put a carrier on 144.490 MHz.

When running wspr tracker on either Atmega328p or ESP8266 with this library and set to:
`#define WSPR_FREQ       144490550UL // 2 mtr`

It does not output on the wanted frequency. Monitoring the spectrum on an SDR shows that it transmits, I see signal on 140.772 MHz but with wider shift as default for wspr so that might be a harmonic from a lower frequency. 

Tested all HF wspr frequencies up to 10 mtr whick works okay.

zanco, 2020-11-22:
I hope you find the time to take a look at this issue. Still no luck to get it to work at 144 MHz.

Thanks,
Ben

mrguen, 2020-12-22:
@zanco @NT7S 

I had the same issue.

It seems div by 4 needs to be activated from 125 MHz upward and not 150 MHz. But when doing so, for 125 MHz the PLL frequency is 500 MHz. If you let it like that the frequencies between 125 and 150 MHz are all changed to 150 MHz.

So a setting that works but is out of specs is

#define SI5351_PLL_VCO_MIN              500000000
#define SI5351_MULTISYNTH_DIVBY4_FREQ   125000000

----------------------------------------
I would also point that Si5351 is supposed to go down to 2.5 KHz from revision 1.0 and that a (OMD) can go to 2048, se personally I have these settings, working, provided I there isn't a bug I haven't seen:

#define SI5351_MULTISYNTH_A_MAX         2048

#define SI5351_MULTISYNTH_MIN_FREQ      320000
#define SI5351_CLKOUT_MIN_FREQ          2500

these two parameters must be set together, it would be best to have

#define SI5351_MULTISYNTH_MIN_FREQ      SI5351_CLKOUT_MIN_FREQ * 128