etherkit/Si5351Arduino #71: Si5351A - drive strength not changing

Marcony35, 2019-11-09:
Hello,

I'm using Si5351A and trying to change drive strenght, but with no effect.
Arduino compiles the code, everything is fine, but on RF meter I don't see the change.

Here is the initialization code for Si5351A...

================================================================
// Initiating the Signal Generator (si5351)
  si5351.init(SI5351_CRYSTAL_LOAD_8PF, 0, 0);
  si5351.drive_strength(SI5351_CLK0, SI5351_DRIVE_2MA);
  // Adjusting the frequency (see how to calibrate the Si5351 - example si5351_calibration.ino)
  si5351.set_correction(CORRECTION_FACTOR, SI5351_PLL_INPUT_XO);
  // The drive strength is the amount of current into a 50Ω load. 2 mA roughly corresponds to 3 dBm output and 8 mA is approximately 10 dBm output (2,4,6 i 8)
  si5351.set_pll(SI5351_PLL_FIXED, SI5351_PLLA);
  si5351.set_freq(vfoFreq, SI5351_CLK0); // Start CLK0 (VFO)
  si5351.set_freq(bfoFreq, SI5351_CLK1); // Start CLK1 (BFO)
  si5351.update_status();
  // Show the initial system information
  delay(500);
================================================================

Is this a bug issue, or maybe IC issue?
Is there any other way to change drive strenght?

NT7S, 2019-11-12:
Unless there was a regression the crept in during a recent update, there shouldn't be a problem with this. I'll try to duplicate the issue using your provided code.

NT7S, 2019-11-12:
I was able to get the expected monotonic increase in output power by changing the drive levels to 2, 4, 6, and 8 mA using a clone Uno and one of my breakout boards. Going to mark this as closed as I don't see a library problem here. Perhaps double-check that you are using the most current version of the library. I suppose it could be a bad IC. It might be worth trying on another one if you have it available.

closed 2019-11-12 by NT7S