etherkit/Si5351Arduino #72: Strange issues with SI5351A-B-GM and ABM8G-25.000MHz-4Y-T3

daspuru, 2019-11-29:
Hi! I am testing the library with the SI5351A-B-GM version of the chip. It's the A version with 8 clock outs. 

I am using a 25Mhz crystal, specifically this one: 
https://www.mouser.mx/ProductDetail/815-ABM8G-25-4YT3

The problem is that when I run a very simple sketch that initializes and sets a freq, I get around 1.5 times the frequency I set. For example in your calibration example, the initial freq I see in the oscilloscope is 15Mhz and it should be 10Mhz, and each time I try to set a freq I see this kind of factor in the output. 

If I hit any key in the calibration sketch, I don't see any changes in the original frequency. 

Do you have an idea where the problem can be?

Thanks!

NT7S, 2019-11-29:
What happens when you try the si5351_example.ino sketch and try varying the output frequency [here](https://github.com/etherkit/Si5351Arduino/blob/master/examples/si5351_example/si5351_example.ino#L38)? Does the output at least change frequency at all?

daspuru, 2019-11-29:
Hi, running the sketch you refer the chip outputs a freq of 21.1Mhz at CLK0 out, and CLK1 is always high. However I tried my original code which changes the output frequency with an encoder and it changes fine, but with this huge factor (around 1.5). My code was using the Adafruit breakout board with the 3 output version.

NT7S, 2019-11-29:
What microcontroller?

daspuru, 2019-11-29:
Teensy 3.2.

I actually want to say that my original code works great with the Adafruit breakout board, all I2C communications seem to be working fine.

daspuru, 2019-11-29:
I actually have two Si5351A-B-GM mounted on my board and both behave in the exact same way, they just have different I2C addresses but the behavior and frequencies outputs the same results.

NT7S, 2019-11-29:
I think it's highly likely that the issue stems from the Teensy, not the Si5351A ICs. I know that the Teensy uses a different toolchain and libraries than most Arduino compatible controllers, and I suspect that's where the bug lies. Saw something similar with the ESP8266. Unfortunately I don't have a Teensy on hand to confirm this. I'll leave this issue open for now, in case I get a chance to get one in the future and try to solve this problem.

Otherwise the only think I can think of that you can do is possibly confirm this by trying to drive your two Si5351A-B-GM with a different controller (at least temporarily) in order to confirm it's a controller problem.

daspuru, 2019-11-29:
Ok. If I change the PLLA freq I got fairly precise values, for example this code outputs 10Mhz and if I change the freq using the same PLL freq seems to be working fine. 

`si5351.set_pll(118000000000ULL, SI5351_PLLA);`

`si5351.set_freq(1000000000ULL, SI5351_CLK0);`

Do you think it can be some kind of library versioning or something like that regarding the comments you say about the microcontroller? As I said I have a production working software using Teensy 3.2 in several boards using the Adafruit breakout boards, Teensy is super stable but I have some ESP32 here, I will test with one of them. 

Thanks a lot!!!!

NT7S, 2019-11-29:
>As I said I have a production working software using Teensy 3.2 in several boards using the Adafruit breakout boards, Teensy is super stable but I have some ESP32 here, I will test with one of them.

Ah, I misunderstood then. I wouldn't try with the ESP32, there's been reports of that having issues as well, which I also need to attend to soon. I'm not sure what would be different in the 8-output version of the chip that would cause this problem, but there might have been some subtlety that I missed. I have used the library with the B and C variants with 8 outputs and it works fine there.

Maybe try a clean installation of the library and try again, just to be sure. Meaning delete the library from Library Manager, clean out the package in the .arduino15/staging/etherkit folder, and then re-install fresh. Let me know what you find.

daspuru, 2019-11-29:
Definitely I am improving. I detected that my voltage was dropping a little bit, so I am now testing with an independent power supply for both Si5351, and now I am getting the correct frequencies :) however, for clocks 7 & 8 (SI5351_CLK6 and SI5351_CLK7) I am not able to set the correct frequencies, for example, if I set:

`si5351.set_freq(1000000ULL, SI5351_CLK6);`

The clock just goes low (no oscillating).

And if I set: 

` si5351.set_freq_manual(1000000ULL, SI5351_PLL_FIXED, SI5351_CLK6);`

I see the output as 48.075 Khz, which should be 10Khz. 

Do you know what can be causing this issue? Again, THANKS a lot!

NT7S, 2019-11-29:
The CLK 6 and CLK7 problems could very well be a bug in my library code. I'll have to investigate here and see what I can find.

daspuru, 2019-11-29:
Thanks Jason, 

I can help to find the bugs if you have a spot on what can be the issue.

Best regards!

daspuru, 2019-12-01:
Due to the problems discussed in this thread regarding the SI5351A and as you said you tested with the SI5351B, with the aim of discarding an issue with this specific A version, I have just tested with a SI5351B, using a breakout board I had brought from China before. The same program has similar problems with CLK6 and CLK7 outs.

`si5351.set_freq(1000000ULL, SI5351_CLK6);`

With the previous code I just see the output six Low, no oscillating.

And with this code:

`si5351.set_freq_manual(1000000ULL, SI5351_PLL_FIXED, SI5351_CLK6);`

I see a similar freq as with the SI5351A, but instead of 48.075Khz I see 52.08Khz. 

For CLK0 through CLK5 I am obtaining the correct frequencies, in the same way as with the SI5351A.

So it seems that the CLK6 & CLK7 problems might be a generalized problem?

Again, thanks a lot!