etherkit/Si5351Arduino #73: Erratic frequency Si5351A Arduino Uno

realbird616, 2019-12-20:
Hi,
Just run the Example sketch and the results  is 42.XXXMhz. not the expected 14.XX Mhz
Changing the CLK0 output to 10 MHz
with " si5351.set_freq(1000000000ULL, SI5351_CLK0);" and I will get 10.Mhz.!!
The same with an input of 12 will give 12Mhz.
Between 13 and 29 will give irrelevant values. Restarting at 30 is then OK ??
I have clean re-install libraries and even try from another computer, same results.
If I use the Adafruit sketch I can get all frequencies normally from A or B on that board.

NT7S, 2019-12-20:
Are you using some kind of breakout board or is this a custom circuit?

realbird616, 2019-12-20:
Hi,
No breakout board. Si5351 connected directly to Arduino. I think it work only once displaying 14MHz ?

kf4mot, 2019-12-20:
Are you sure you're not just seeing the 3rd. harmonic?

realbird616, 2019-12-20:
I can’t say, I dont have a spectrum analyzer but it looks as a multiple of the requested 14 Mhz. With the Adafruit sketch no problem getting 14 Mhz on my HP freq.counter.

kf4mot, 2019-12-21:
Do you have a hf receiver you can try to listen in on?
I don't think I'd trust a counter to show the fundamental of a square wave. If you can add lpf put in line with the counter that may help.
That doesn't explain it working with the Adafruit code, but that could just be a coincidence with your test setup.

realbird616, 2019-12-21:
You are absolutely right. I was relying blindly on this HP5316A. With a 2pf it’s now showing the fundamental  at 14.Mhz. I should have tried my 30 years old Fluke that was right on with no filtering. Tks again

closed 2019-12-21 by NT7S