etherkit/Si5351Arduino #75: It would be nice to easily have CLK0 and CLK1 in phase....

frohro, 2020-08-05:
Hi,
Ideally, I would like CLK0 to go at twice the frequency of CLK1, but have them in phase.  That way I can drive the low and high order control bytes of a 4x1 analog MUX for doing SDR to end up with I and Q. (It counts 00, 01, 10, 11,..., where the low order bit is half the frequency of the high bit.)  I think the restrictions that occur with a 90 degree phase delay between CLK0 and CLK1 do not occur with this setup because the phase delay is zero.  It would be nice to let the library deal with the intricacies though.  ;-)  This scheme is used by [Guido in the UCX-SSB](https://github.com/threeme3/QCX-SSB), but he isn't using your handy library.  I would like it to be easy for those who prefer the library route.
Thanks & 73,
Rob
KL7NA

NT7S, 2020-08-06:
Hello,

Have you tried setting each CLK output to the same phase with the set_phase method after setting the frequency? If I understand what you are trying to do correctly, that should be what you are looking for. From my experiments in trying to generate PSK with the Si5351A, I do know that you can set both outputs to the same phase as long as you don't change the frequency again. Once you do, you have to reset the phase (and there's a discontinuity in the waveform).

Better yet, you might want to set CLK0 as desired, and then set CLK1 to get its clock from CLK0 and then use the set_ms method and use the R_DIV divider of CLK1 to divide by two. I should think that would guarantee phase fidelity, but I haven't tried it to verify that. I'm going to close the issue on the assumption that one of the two methods will work for you, but if not, feel free to comment some more.

closed 2020-08-06 by NT7S

la3pna, 2020-08-07:
Going to be a PITA and open this issue again for a comment:

Option 2 as outlined by Jason works and are tested here.

FWIW, you can't have anything be in phase and on different frequencies at
the same time. What you want is to have the leading edges synchronized or
clocked.

BR.
Thomas.

frohro, 2020-08-09:
> CAUTION: This email originated from outside the Walla Walla University
> email system.

Hi Thomas,

Any chance you could provide me an example?

Thanks,

Rob

frohro, 2020-08-09:
I really think the idea of setting the output divider as divide by two for the slower clock is what I need to do, but I can't figure out how to get the struct Si5351RegSet ms_reg needed for the set_ms method.  An example would be best, but if I could even figure that out, it might solve my entire problem.
Thanks!
Rob