etherkit/Si5351Arduino #77: Compatibility

scoopgracie, 2020-10-28:
Is this compatible with https://www.adafruit.com/product/2045?

la3pna, 2020-10-28:
Buy this instead:
https://www.etherkit.com/rf-modules/si5351a-breakout-board.html
Then you are supporting the developer of the project!

ons. 28. okt. 2020 kl. 03:52 skrev Samuel Sloniker <notifications@github.com

>:

> Is this compatible with https://www.adafruit.com/product/2045?

scoopgracie, 2020-10-28:
Okay, thanks! What is TCXO?

la3pna, 2020-10-28:
Temprature Compensated Xtall Oscillator, it means that the reference
oscillator used is more stabile and accurate than a normal crystal.

ons. 28. okt. 2020 kl. 14:29 skrev Samuel Sloniker <notifications@github.com

>:

> Okay, thanks! What is TCXO?
>
>
> > Buy this instead:
> > https://www.etherkit.com/rf-modules/si5351a-breakout-board.html
> > Then you are supporting the developer of the project!
> >
> > ons. 28. okt. 2020 kl. 03:52 skrev Samuel Sloniker <
> > notifications@github.com
> > >:
> >
> > > Is this compatible with https://www.adafruit.com/product/2045?
> > >
> > > <
> >
> > >
> > >
> >
> >
> > With Best regards, Thomas S. Knutsen.
> >
> > Please avoid sending me Word or PowerPoint attachments.
> >
> > <
> https://github.com/etherkit/Si5351Arduino/issues/77#issuecomment-717746481
> >,
> > <

scoopgracie, 2020-10-28:
Okay. For amateur radio use, would the regular version be sufficient?