etherkit/Si5351Arduino #85: need definition of error messages

steeley, 2021-10-16:
I'm using this library with a 4351b, 27Mhz Xtal and it seems to work well.

However I would like to see what LOS and REVID actually mean. in my case they always show 1
'indicating some kind of error"?? as show below. but my device seems to be working....

PLLA: 800000000   PLLB: 800000000  SYS_INIT: 0  LOL_A: 0  LOL_B: 0  LOS: 1  REVID: 1

Thanks

evg2, 2023-06-22:
I have same issue with Si5351, 25MHz and work well.
I do not know what mean SYS_INIT: 0 LOL_A: 0 LOL_B: 0 LOS: 1 REVID: 1.
I know that must be SYS_INIT: 0 LOL_A: 0 LOL_B: 0 LOS: 0 REVID: 3.

Thanks

tcarney57, 2023-06-22:
LOL means "loss of lock" (of the PLL), and LOS means "loss of signal" of the crystal or external clock. There's a few registers for these per generated clock. One set is "sticky" (meaning they have to be deliberately cleared), and the other is transient. These can be read from the device and used as desired. The library example sketch <si5351_example.ino> shows their use. In this case, just to confirm the device is working properly.

For detailed information on these and other registers, see Silicon Labs/Skyworks "AN619: Manually Generating an Si5351 Register Map."

evg2, 2023-06-22:
What is REVID doing. Why changes from 1 to 3 if it's normal?

NT7S, 2026-11-07:
Fixed in [v2.4.1](https://github.com/etherkit/Si5351Arduino/releases/tag/v2.4.1) (PR #110), hardware-verified on Si5351B and Si5351C. See the release notes for details; please reopen if you still see this on v2.4.1.

closed 2026-11-07 by NT7S