etherkit/Si5351Arduino #87: Si5351 - ESP8266 WSPR beacon VHF

iz2fly, 2022-06-08:
I am building this very simple project to transmit in WSPR in the 2m band with ESP8266 and Si5351.
See web page: 
https://antrak.org.tr/blog/esp-wspr-simple-and-inexpensive-wspr-transmitter/
sketch:
https://antrak.org.tr/wp-content/uploads/ESPWSPR-yazilimlar.zip

libraries:
// Etherkit Si5351 (Library Manager)
// Etherkit JTEncode (Library Manager)
// Time (Library Manager)
// Wire (Arduino Standard Library)
// NTPtimeESP
// ESP8266WiFi

I solved the problem of setting the frequency by changing the variable freq:
unsigned long freq         = 144489000UL
unsigned long long freq = 144489000ULL
otherwise the calculations were incorrect.

But now there are continuous clicks every half second or so, this does not allow the signal to be decoded. (In 144MHz)
In HF no problems.
Ideas? Thanks

Bleach665, 2023-04-19:
Same issue.

iz2fly, 2023-04-20:
Now I'm trying with an RS41 radiosonde that drives the Si5351 with the serial port, I added a 10MHz OCXO to stabilize the frequency and the decoding is almost complete. In 144MHz however, the dirty signal quality remains.
Here you find the software.
https://github.com/mikaelnousiainen/RS41ng