etherkit/Si5351Arduino #92: Problems with SI5351B-GM1 only two outputs working and they don't match.

zindello, 2023-05-02:
**Describe the bug**
I am having strange issues when using the SI5351B-GM1

**Setup**:
Microcontroller: Seedstudio Xiao ESP32-C3
Crystal: TXC 7M-27.000MEEQ-T (Per the recommended crystals)
Si5351 Chip: SI5351B-GM1 (16QFN)

**To Reproduce**
Using the supplied library, I can init the library and communicate with the chip, however when setting outputs strange things happen.

Setting an output on CLK0 does nothing
Setting an output on CLK1 provides output on the CLK0 pin
Setting an output on CLK2 provides an output on the CLK1 pin
Setting an output on CLK3 does nothing
Setting an output on CLK4 does nothing (Given there is no CLK4 this is expected)
Setting an output on CLK5 does nothing
Setting an output on CLK6 does nothing
Setting an output on CLK7 does nothing

**Expected behavior**
CLK0-CLK3 all output as expected.

**Additional context**
I feel like this might be the result of the 16QFN chip having it's registers configured a little differently and as such the library isn't setting the registers properly? It would be neat if I could use this library to set the registers on this chip as I really don't want to write a library to do what this one does myself. Im going to dig into the datasheet and the library to see if I can figure out what's going on.

zindello, 2023-05-03:
Ok, further to this I think I've found the problem.

The "A" version of the chips all have registers for 8 clocks, whether they use them or not, so the library works for all of those versions.

The "B/16QFN" version of the chips however have four clock outs, but the registers are in different places, mostly however within the same address spaces.

Why use the 16QFN version? They're roughly half the cost of the 20QFN chips.

So far I've been able to establish that.

CLK0 is in the place where CLK1 is on the other versions - This works if you use CLK1 in the library
CLK1 is in the place where CLK2 is on the other versions - This works if you use CLK2 in the library
CLK2 is in the place where CLK5 is on the other versions - This works if you use CLK5 in the library
CLK3 is in the place where CLK7 is on the other versions - However this doesn't work if you use CLK7 in the library.

So it looks like everything works except for CLK3. I had a look at the library and it looks like you're using binary maths to figure out the location of a bunch of the registers, bit shifting the clock number to hit the right register value to set the register you need to for each clock. Unfortunately for the B revision chips this won't work as the registers are in strange places. See attached screenshots of the data sheets.

https://www.skyworksinc.com/-/media/Skyworks/SL/documents/public/data-sheets/Si5351-B.pdf

[image]
[image]

ajsoukey, 2024-12-13:
Any chance you have found a fix for this? I didn't realize that buying the 16QFN wouldn't actually get me all 4 ports available. If not, I guess the next board version will have to be the 20QFN.

NT7S, 2026-11-08:
Still open: confirming this needs a bench test on an Si5351B-GM1 (16-QFN) part, which we don't have on hand yet. If anyone with a GM1 can report which physical pin each of CLK0–CLK3 drives with v2.4.2, that would settle it.