etherkit/Si5351Arduino #93: Handle missing external clock, either fallback to internal oscillator or switching off output

pignoniG, 2023-07-07:
Hi, is there any recommendation or plan on implementing automatic switch between external and internal reference?
It would be nice to be able to turn off the output if no reference is connected or switch to the internal one.
Thanks

la3pna, 2023-07-07:
This will require external circuitry.
Easiest approach is a diode detector on the external input, and if the
level is satisfactory you put a pin on the microcontroller high. Then the
program needs to take into account this and switch.
Has been done and works ok.

BR
Thomas

pignoniG, 2023-07-07:
Hi Thomas, I can reliably “sense” the ext reference but didn’t manage to
reliably reconfigure the ic wile it’s running. Do you have any example
code?
Thanks