etherkit/Si5351Arduino #94: VCXO Calibration Issue

M0XFX, 2023-09-27:
Hi I am using the SI5153B to modulate narrow FM which it is working quite well.

I have used the frequency correction procedure and it works well across a wide range of frequencies. 
However; the calibration goes out of whack when I invoke the set_vcxo() function.

I carried out the calibration using PLLB and I cant figure out why this is happening. Any Ideas?

#define PLLB_FREQ    87600000000ULL

si5351.init(SI5351_CRYSTAL_LOAD_8PF, 0, 0);

si5351.set_vcxo(PLLB_FREQ, 40);     // as soon as I add this line it goes slightly off frequency and gets worse through the spectrum. I have set the ppm min to 0 in the header file and when I set the pull range to 0 when calling the function it corrects the frequency offset issue but then obviously  I can't modulate FM. 

//si5351.si5351_write(SI5351_VXCO_PARAMETERS_LOW, 1000); // Not sure what this is for

si5351.set_ms_source(SI5351_CLK2, SI5351_PLLB);  

si5351.set_correction(-14468, SI5351_PLL_INPUT_XO);
 
Not sure if this is a bug or just something I am missing. Thank you in advance for your help

BTW im using the latest library in github and there is no voltage a the VC pin of the IC. 

Regards John M0XFX

obroad, 2024-07-15:
When you say no voltage at VC are you biasing it to VS/2?
I'm not familiar with this device but as far as I'm aware a VCXO gives its nominal frequency for VS/2, so leaving it grounded would give its maximum offset.

M0XFX, 2024-07-15:
Hi, thank you for your comment.

Yes you are correct, since writing this post I figured it out and I forgot to update the issue.

The VC pin need to be biased at VCC/2 to produce nominal frequency.

Also the issue with stability was down to a calibration issue.

I will close the issue, 

Apologies for not taking it down sooner.

Regards

John

closed 2024-07-15 by M0XFX