etherkit/Si5351Arduino #96: Si5351.h and AtTiny3216

WaynePauly, 2023-12-17:
The library does not seem to work with the new Microchip AtTiny's, at least with the AtTiny3216, but the Adafruit 5351 library does. I am using Arduino 1.8.15 and MegaTinyCore and the supplied example. The code compiles and runs but the chip does not respond. I have used the library many times before with Nano's and ESP8266's with great success and hoped to port our code to the tiny processors for our balloon trackers.

SpenceKonde, 2024-01-04:
I maintain megaTinyCore, and if you have questions on what's going on once you've localized the problem I would be happy to advise.

NT7S, 2024-10-11:
Thank you, I recently acquired a ATtiny3226 to use with megaTinyCore and this library and I saw the same issue. It has been corrected in v2.2.0. The problem was with my `Wire.requestFrom` call using the wrong parameter to keep the bus, instead of releasing it. That's apparently not a problem with the older micros, but it won't work on this family (and I suspect it causes problems with other families of micros as well).

@SpenceKonde Thanks for the megaTinyCore, it's obvious you have put a lot of work into it.

closed 2024-10-11 by NT7S

SpenceKonde, 2024-10-17:
Great, glad to hear it!