etherkit/Si5351Arduino #97: The comments in the title block of SI5351.set_freq(freq, clk) say freq units are Hz.  I think they are Hz*100ULL.

frohro, 2024-04-06:
This is on line 180 of si5351.cpp.  I believe the same issues occur with SI5351.set_freq_manual and probably other functions should be checked.  This is important as VScode and other IDEs display those comments when you hover over the function call.

NT7S, 2024-04-09:
Thank you, I'll fix this in the next update, which shouldn't be too long now that I've got a lab again.

conr2286, 2024-12-05:
I've bumped into this one too and just changed the comment re. set_freq() params in my clone.  It was a puzzle for a moment as I was sure the dividers must be set wrong. 🙃

Let me know if u want a pull; I'll make the fix on a clean fork (you don't want the hacked-up bits I use;).

JimC, KQ7B

NT7S, 2026-11-07:
Fixed in [v2.4.1](https://github.com/etherkit/Si5351Arduino/releases/tag/v2.4.1) (PR #110), hardware-verified on Si5351B and Si5351C. See the release notes for details; please reopen if you still see this on v2.4.1.

closed 2026-11-07 by NT7S