etherkit/Si5351Arduino #101: Can't seem to generate requested frequency

retrorepair, 2024-11-07:
I've altered the example ino to specifically output 4.43548375mhz by changing:

Si5351.set_freq(443361877ULL, SI5351_CLK0);

Is that not going to work for some reason? I can't seem to find documentation on this

retrorepair, 2024-11-07:
Should the serial monitor show anything other than 0 for all registers? Even with the default unedited ino I get this

retrorepair, 2024-11-09:
I corrected the frequency to 4433618750ULL but still not working

la3pna, 2024-11-09:
What device are you running on? How is the I2C connected? does it have
propper bias resistors?
 With a SI5351 or one of the clones?

la3pna, 2024-11-09:
Frequencies are given in 0.01Hz as the input. Your input will give
44.336MHz output.
https://github.com/etherkit/Si5351Arduino

retrorepair, 2024-11-09:
Everything is set up fine and the board is a nano. The adafruit library works but I can't deal with the method they use to generate a frequency which I assume is closer to the register setup than the etherkit library